mattairtech/ArduinoCore-samd #2: Support for platformio?

jarzebski, 2016-12-08:
Hi, did you plan support platformIO ? Will be great!  http://platformio.org/boards ?

mattairtech, 2016-12-12:
Yes, I will support PlatformIO, though I won't have time for this until after I release the next arduino core sometime in January.

joseangeljimenez, 2017-05-19:
I am willing to contribute/help you to support platformIO for your boards definition. In case you are OK with it... How can I help you?

glennsky, 2018-06-30:
While for platformio & CAN bus support for the SAMC21 - I stumbled upon this repository:
[https://github.com/hugllc/samc21_can](url) 

Which has a fork of the platformio platform-atmelsam that works with SAMC21.
[https://github.com/prices/platform-atmelsam](url)
and a fork of ArduinoCore-samd
[https://github.com/prices/ArduinoCore-samd](url)

Perhaps this could help with getting the mattairtech ArduinoCore-samd working with platformio?!?

mr3188, 2019-10-14:
Any advance on this enhancement? I would also like to help if needed.