mattairtech/ArduinoCore-samd #3: Support for ATSAMD11C14

sandeepansengupta, 2017-02-19:
It's not clear if your core supports AtSAMD11C14 MCU (14 pin variant). Adding it will be great.

mattairtech, 2017-02-21:
I have added a new board variant to support a generic D11C14A board. It will be available along with the next core update (hopefully within a couple of weeks). Here is the planned pin mapping:
```
====================================== ATsamD11C14A =====================================
Other  COM   PWM  Analog  INT  Arduino*           Arduino*  INT  Analog  PWM   COM  Other
=========================================================================================
                                   -------------------
     RX2*    TCC01    *    *    5 | A5             A4 | 4    *    *     TCC00  TX2*   REF
 TX1 / MOSI  TCC02         *    8 | A8             A2 | 2    *    *                   DAC
 RX1 / SCK   TCC03         *    9 | A9            Vdd |
     SDA     TC10     *   NMI  14 | A14           Gnd |
     SCL     TC11     *    *   15 | A15           A25 | 25                    USB/DP
BOOT                           28 | A28/RST       A24 | 24                    USB/DM
SWDCLK  MISO                   30 | A30           A31 | 31   *                SS    SWDIO
                                   -------------------

* When using ONE_UART_NO_WIRE_ONE_SPI, Serial1() refers to TX2/RX2 instead of TX1/RX1.
* Tone available on TC2. TC2 is not routed to pins with the D11C14A.
```

sandeepansengupta, 2017-02-22:
Thank you sir, thanx a lot.
Is there any way to use tone()
And also do you have anything in mind for VCP-MSD/CDC-MSD composit bootloader

sandeepansengupta, 2017-02-22:
Just a quary, why not use 0-12 pin numbering for arduino as there will be a new core. Is it in conflict with something else ?

And again, thank you for making these cores, specially ATSAMD21E18

mattairtech, 2017-02-25:
tone() will be supported on TC2 (note that TC2 is not routed to external pins on the ATsamD11C14A). Currently, I have no plans to add CDC or VCP to the MSD bootloader (maybe if there is more interest). The user can either use sequential 0-12 pin numbering (PIN_MAP_COMPACT defined) or use numbers that correspond to the port pins (ie: A7 = Arduino pin 7; PIN_MAP_STANDARD defined). PIN_MAP_STANDARD is the default and is what is shown in the ASCII pinout above.

SandeepanSengupta, 2017-03-14:
@mattairtech , is it (SAMD11C14 arduino Core) releaded yet ?

mattairtech, 2017-03-15:
I think it will be released by next Friday, March 24.

closed 2017-03-16 by SandeepanSengupta