mattairtech/ArduinoCore-samd #4: Weren't you supposed to release the new core by now?

mattairtech, 2017-03-25:
OK, I was planning on releasing the new core by Friday March 24. The new release date might be in 1 week. I ran into problems with the PLL on the L21. The PLL works fine on the D21, C21, and probably the D11s (will test soon), but it does not function at all on the L21. The PLL will be used to generate 96MHz (then divided by 2 for Fcpu; TCCs can use the 96MHz) from either crystal (32.768KHz or any high speed crystal between 400KHz and 32MHz). There is also the option to use an internal oscillator, which can be calibrated to the USB SOF signal (except C21). Anyway, since I am giving up on the L21 PLL, it will instead use the DFLL for the 32.768KHz crystal with the L21 only (use of the high speed crystal may not be supported). I will post daily updates here.

mattairtech, 2017-03-25:
The new bootloaders have already been installed on products that have shipped. These bootloaders use the same clock code as the new core (with the 4 clock source options), but they were configured to use the internal oscillator (with USB calibration on all but the C21). The D21/D11 boards will work with the existing core, but the L21 requires a recompile of the bossac command line tool which is used by the Arduino IDE to program the FLASH. I have already compiled for Linux, but Windows and OS X will take longer. I will work on that today (3/25/2017).

mattairtech, 2017-03-25:
OK, figured out the problem with the L21. I had not set the performance level to PL2 (highest). The D21/D11/C21 do not have performance levels. I knew that it existed, but I had assumed that it would be set to PL2 by default. The PLL works fine with both crystals. Now I need to get back to compiling bossac for Windows and OS X.

mattairtech, 2017-03-30:
Bootloaders are tested and now available. Bootloader utility is compiled for Linux (64bit, 32bit), Windows (64bit, 32bit), and OS X (64bit). Now finishing updates to new CMSIS. Tomorrow I will finish merging latest changes from upstream ArduinoCore-samd. I will also add an external interrupt speedup from the issue tracker.

jgilbert20, 2017-04-02:
Thanks for working on this! :)

SandeepanSengupta, 2017-04-09:
Any news about _ATSAMD11_**C14** Core/Board and Bootloader ?

mattairtech, 2017-04-11:
I have completed testing of the core with the SAMDs, but I am currently hung up on the L21 testing. If I cannot get the L21 to work, I will put it on hold and release the core with D21, D11, and possibly C21 support only. I should have more information about the fate of the L21 in about 12 hours or so. Right now I am having great difficulty getting the external interrupt controller to work.

mattairtech, 2017-04-12:
Still working on interrupts with the L21 (and C21). I had reduced the code to a minimum, checked the disassembly (including all instructions, constant values, and addresses), cross checked with ASF code, and it still won't work. The peripheral is almost identical between all three chips, so I was not expecting these problems. I finally compiled an ASF example and it finally works. Now I am trying to find the magic code is ASF. I should find this today and get on with testing the ADC, DAC, PWM, SPI, I2C, and serial.

mattairtech, 2017-04-12:
OK, I think I am past the external interrupt problem. The EIC.CONFIG register is enable-protected in the L21 and C21. The EIC must be disabled each time the SENSE bits (or filter) are changed. On to the ADC.

mattairtech, 2017-04-14:
I had to spend a little more time with interrupts. For some reason, pin PA31 cannot be used with CHANGE or RISING interrupt modes on the L21 (and probably C21), but it works fine on the D21/D11. The analogWrite() function will exit without doing anything if the user specifies this pin with one of these modes. PWM was next. I was able to get the TCC timers working, but I am still working on the TC timers.

mattairtech, 2017-04-15:
OK, all hardware testing is complete. Everything works. The L21 has two DACs and both are supported. The C21 has 2 ADCs and both are used. I tested SPI by listing the directory contents of a MicroSD. I tested I2C with an RTC chip.

I still have to:
Arduino version testing / different platform testing / some additional USB testing: a few hours
add the remaining generic board variants: a few hours
documentation and release: several hours
The remaining testing could have surprises that could add time.

joseangeljimenez, 2017-04-15:
Thanks a lot for your great work!

jgilbert20, 2017-04-15:
Looking forward to it!!

jgilbert20, 2017-04-19:
Hey Justin, any chance you would consider pushing our your current work to a branch? I'm struggling to get my clock-less configuration working reliably across different chips.

mattairtech, 2017-04-20:
I was delayed by taxes over the weekend until Tuesday. I had to catch up on other work, but I am back full time on this now. This will be released Friday, regardless of the state of the documentation.

mattairtech, 2017-04-22:
Released!
I still have more documentation left.
Also, the Windows version of IDE 1.8.2 doesn't seem to be sending the auto-reset signal (or my board is not getting it). So, auto-reset is not working. Manually enter the bootloader for now by double-tapping the reset button. Auto-reset works fine on Linux. Maybe I'll figure it out over the weekend.

mattairtech, 2017-04-24:
Fixed auto-reset not working on some versions of Windows (mine was Windows 7 64-bit). I also worked on the documentation and fixed the formatting.

SandeepanSengupta, 2017-04-24:
Thank you for releasing the core @mattairtech
👍

glennsky, 2017-04-24:
Yes, thanks...   do you have an update for:
https://www.mattairtech.com/software/arduino/package_MattairTech_index.json

to be able to load this with the Arduino boards manager?

mattairtech, 2017-04-24:
Use the Beta branch with this core. That is, point your Arduino IDE to https://www.mattairtech.com/software/arduino/beta/package_MattairTech_index.json. See the Beta Builds section in the root README.md for more info.

glennsky, 2017-04-24:
Great - beta installs fine.
Thanks.

joseangeljimenez, 2017-05-19:
Hi @mattairtech, I guess you can close this issue! Thanks a lot for your great work.

mattairtech, 2017-10-07:
OK, finally released the release version, so closing.

commit d9df9083 (2016-05-29): Fixed cortex-M hooks for RTOS

Fixes #4
closed 2017-10-07 by mattairtech