mattairtech/ArduinoCore-samd #8: Tools Menu Options

perigalacticon, 2017-09-15:
Hello I am trying this core for the first time.  I don't see most of the options that should be in the tools menu like:  "Bootloader Size Menu":

[image]

Is there something wrong with my installation?  I am using IDE 1.8.4.

mattairtech, 2017-09-16:
Please use the Beta for now for all of the new menus (see https://github.com/mattairtech/ArduinoCore-samd/README.md). I will finally make the next beta version (which should be released in just over 1 week) the new release version as well. I intend to have more release versions from then on.

mattairtech, 2017-10-07:
This is now in the release version. The beta track always includes the release versions as well.

closed 2017-10-07 by mattairtech