mattairtech/ArduinoCore-samd #9: Dataflash library compatibility error

perigalacticon, 2017-09-18:
I would like to use a library for interfacing with Dataflash, D series SPI Flash memory with the SAMD11.  I am successfully using the "arduino-dataflash" library  (by Mooz from https://github.com/BlockoS/arduino-dataflash ) with ATTINY841 boards but for the SAMD11 the compiler gives the following error:

`C:\Program Files (x86)\Arduino\libraries\DataFlash\DataFlash.cpp: In member function 'void DataFlash::setup(int8_t, int8_t, int8_t)':

C:\Program Files (x86)\Arduino\libraries\DataFlash\DataFlash.cpp:152:15: error: 'SPSR' was not declared in this scope

 m_SPSR = (SPSR & SPI_2XCLOCK_MASK);

           ^`

I am using the Generic D11C14A board option in the IDE.

perigalacticon, 2017-09-18:
The relevant code from DataFlash.cpp is:

`    /* Save SPI registers with settings for DataFlash for fast restore
     * in .begin(). */
    m_SPSR = (SPSR & SPI_2XCLOCK_MASK);
    m_SPCR = SPCR;

    /* Clear pending SPI interrupts */
#ifdef __AVR__
    uint8_t clr;
    clr = SPSR;
    clr = SPDR;
    (void) clr; // Prevent variable set but unused warning. No code generated.
#endif
`

perigalacticon, 2017-09-18:
`;oi;jn`

perigalacticon, 2017-09-18:
I think the problem may be with the definition of "SPI_2XCLOCK_MASK", I found it in Arduino file "SPI.h":
https://github.com/arduino/Arduino/blob/master/hardware/arduino/avr/libraries/SPI/src/SPI.h
`
#define SPI_2XCLOCK_MASK 0x01 // SPI2X = bit 0 on SPSR
`

mattairtech, 2017-09-20:
It looks like the library is compatible with the AVR core, but not the SAM or SAMD cores. While there is an "#ifdef __AVR__"  in at least one place, there are other places in the code that reference the SPSR and SPCR registers, which do not exist with SAMD. It looks like the dataflash library mostly uses the SPI library, but it directly accesses these registers (looks like only once), breaking compatibility for non-AVR. This should be pretty easy to fix, but you will have to contact the author. It looks like a performance enhancement that should not be needed for the SAMD.

closed 2018-04-03 by mattairtech