mattairtech/ArduinoCore-samd #14: sam.h file missing

dunk8888, 2018-01-23:
Hi,how do i get rif of the sam.h error, its looking for that file and not finding it.

Is there a core for amd witch has samd 9-10-11-21 etc or do i need to find cores for them seperatly,ile be messing around with some of these mcu as i want to move up from avr. Ime fairly new to programming and ime allways getting errors.I just need a good starting point so i can start using some of these mcu,mainly samd 11 and 21.Ime using arduino ide 1.8.5 cheers

mattairtech, 2018-01-24:
Did you install this using Boards Manager in the Arduino IDE? The sam.h file is a part of the CMSIS-Atmel package (https://github.com/mattairtech/CMSIS-Atmel) which is installed automatically when using Boards Manager. It should be installed in <your localappdata path>/Arduino15/packages/MattairTech_Arduino/tools/CMSIS-Atmel/1.0.0-mattairtech-2. There are other dependencies that are installed as well. Also, boards manager might not install/uninstall the core or tools properly if the contents of the arduino15 directory has been manually modified, so be sure to delete *all* manually installed/created folders (not just files), then try re-installing using Boards Manager, and make note of any errors.

mattairtech, 2018-01-24:
To answer your second question, the D9 and D10 are subsets of the hardware present on the D11 (for example, no USB), and the D20 is a subset of the D21. It is far more likely that engineers/companies will buy these chips rather than enthusiasts, since the price difference is not much when buying single chip quantities or development boards. It should not be too much trouble for an engineer to modify the existing cores to support the D9, D10, and D20, but I wouldn't recommend these chips to enthusiasts.

closed 2018-04-03 by mattairtech