mattairtech/ArduinoCore-samd #19: Can't install beta versions

obdevel, 2018-06-14:
I'm trying and failing to update my boards manager to install the beta versions, to support my own board based on SAMC21G18A (for CAN bus support). I have other cores successfully installed (MightyCore, TinyCore, ESP8266), as well as 1.6.16 of this core, so the process should work. I'm running Arduino IDE 1.8.5 on macOS 10.13.5.

I add the URL to the URL list in Preferences but nothing new appears in the boards manager. I can access the beta JSON file at the given URL (using curl).

Any thoughts ? Any log files I can look at ? Thanks

mattairtech, 2018-06-14:
Are you using the main beta json (https://www.mattairtech.com/software/arduino/beta/package_MattairTech_index.json), rather than an individual beta version json? Go to File->Preferences and place a check in the "Show verbose output during compilation" box, though this might not generate any additional error messages (the IDE should warn about certain JSON file problems). Also, be sure that there are no manual MattairTech installations in the Arduino15 directory (completely delete any old MattairTech core directories in this case).

obdevel, 2018-06-14:
I think the beta link in the README is out of date. I found the https://www.mattairtech.com/software/arduino/ directory and that seems to have everything I'll need. So I now have the 1.6.18-beta-b0 core installed in boards manager now, and can select Xeno Mini and SAMC21G18A.

Just need to install the tools to solve these errors:

Index error: could not find referenced tool name=bossac version=1.7.0-mattairtech-3 packager=MattairTech_Arduino
Index error: could not find referenced tool name=CMSIS-Atmel version=1.0.0-mattairtech-2 packager=MattairTech_Arduino
Index error: could not find referenced tool name=bossac version=1.7.0-mattairtech-3 packager=MattairTech_Arduino
Index error: could not find referenced tool name=CMSIS-Atmel version=1.0.0-mattairtech-2 packager=MattairTech_Arduino

Any help installing these in the correct places would be a great help.

Thanks

mattairtech, 2018-06-14:
I just checked the README link (same as the one I posted above) and the link seems fine (I checked the file contents, and I have been using it on my own computer as well for a while). If you have more than one URL listed in "Additional Board Manager URLs", please use a newline to separate them. For a manual installation, all of the files you need are in the https://www.mattairtech.com/software/arduino/  directory. Installation requires some folder renaming, so that the directories are like so:
arduino15/packages/MattairTech_Arduino/tools/bossac/1.7.0-mattairtech-3/bossac(.exe)
arduino15/packages/MattairTech_Arduino/tools/CMSIS-Atmel/1.0.0-mattairtech-2/CMSIS/Device/ATMEL/<MCU folders>

obdevel, 2018-06-15:
It seems that the release and beta packages don't like to co-exist. If I remove the release package, the beta installs fine. If I add back the release package, the beta 'disappears' from boards manager. This is ok as I just need the beta core for SAMC21xxx.

Any ideas for burning the bootloader on to a brand new chip ? I could buy an Atmel-ICE and I am also investigating OpenOCD on my Raspberry Pi. Other options would be welcome. This is my first project that doesn't use ATMEGA or ATTINY. fwiw, I would buy one of your boards, but I'm specifically focused on CAN bus right now.

mattairtech, 2018-06-19:
I think with the next beta release (midweek probably), I will combine the beta releases with the main releases in the main json (so just one URL). As you noted, it doesn't look like they both can be installed at the same time. You will probably need to remove the arduino15/packages/MattairTech_Arduino/hardware/samd/1.6.18-beta-b0 directory manually and re-install using boards manager, but only after ensuring that only one MattairTech json URL is present in Preferences. The installation may be damaged by the problems associated with having both jsons. Note that the Arduino IDE gets all of the information about each package from the arduino15 directory. If you do not delete the files in arduino15/staging/packages, then those will be used instead of downloading new files. It is possible that you will need to delete the entire arduino15/packages/MattairTech_Arduino directory, but I don't think so.

I use an Atmel Ice to program. Programming the D21 and D11 is possible through the Arduino IDE, but not the C21, L21, or D51. I will need to supply a new openocd package, but that won't happen anytime soon. I use Atmel Studio in these cases (unfortunately, no Mac or Linux support).

mattairtech, 2018-06-27:
Recent beta versions are now included in the main release json, so you should be able to use just that.