mattairtech/ArduinoCore-samd #20: Core compilation error - 1.6.18-beta-b0

obdevel, 2018-06-15:
Just trying the Blink sketch to test my installation and I get core compilation errors.  Config is:

[image]

Errors:
```
...
"/Users/xx/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-gcc" -mcpu=cortex-m0plus -mthumb -c -g -Os -Wall -Wextra -std=gnu11 -ffunction-sections -fdata-sections -nostdlib --param max-inline-insns-single=500 -MMD -DF_CPU=48000000L -DARDUINO=10805 -DARDUINO_SAMD_ZERO -DARDUINO_ARCH_SAMD "-I/Users/xx/Library/Arduino15/packages/arduino/tools/CMSIS/4.5.0/CMSIS/Include/" "-I/Users/xx/Library/Arduino15/packages/MattairTech_Arduino/tools/CMSIS-Atmel/1.0.0-mattairtech-2/CMSIS/Device/ATMEL/" -DCLOCKCONFIG_32768HZ_CRYSTAL -DUSB_DISABLED -DONE_UART -DONE_WIRE -DONE_SPI -D__8KB_BOOTLOADER__ -D__SAMC21G18A__  "-I/Users/xx/Library/Arduino15/packages/MattairTech_Arduino/hardware/samd/1.6.18-beta-b0/cores/arduino" "-I/Users/xx/Library/Arduino15/packages/MattairTech_Arduino/hardware/samd/1.6.18-beta-b0/variants/Xeno_Mini" "/Users/xx/Library/Arduino15/packages/MattairTech_Arduino/hardware/samd/1.6.18-beta-b0/cores/arduino/wiring_analog.c" -o "/var/folders/03/s3pv8qk91w78m4_st_nbxl5m0000gn/T/arduino_build_389544/core/wiring_analog.c.o"

/Users/xx/Library/Arduino15/packages/MattairTech_Arduino/hardware/samd/1.6.18-beta-b0/cores/arduino/wiring_analog.c: In function 'analogReference':
/Users/xx/Library/Arduino15/packages/MattairTech_Arduino/hardware/samd/1.6.18-beta-b0/cores/arduino/wiring_analog.c:234:22: error: 'AR_INTREF_1V0' undeclared (first use in this function)
   } else if (mode >= AR_INTREF_1V0) {  // Values starting at AR_INTREF_1V0 are used for the Supply Controller reference (AR_INTREF)
                      ^
/Users/xx/Library/Arduino15/packages/MattairTech_Arduino/hardware/samd/1.6.18-beta-b0/cores/arduino/wiring_analog.c:234:22: note: each undeclared identifier is reported only once for each function it appears in
exit status 1
Error compiling for board MattairTech Xeno Mini.
```

mattairtech, 2018-06-27:
This works for me. I suspect this has to do with your previous installation problems. Do you still have this problem after switching to the new json?

glennsky, 2018-06-30:
The SAMC21 doesn't seem to have a AR_INTREF_1V0 defined (AR_INTREF_1V024 is defined) --- so I added this after line 87 in https://github.com/mattairtech/ArduinoCore-samd/blob/master/cores/arduino/wiring_analog.h

`AR_INTREF_1V0 = 6,`

This gets it to compile, though not sure if the analog would work as expected.

togglebit, 2019-08-18:
FWIW I had to do the same the same thing with Beta-b1. Compiles, not tested analog.

commit 54a0c3de (2015-11-24): [Wire] simplified coding unnecessarily complex (hfvogt)

In the wire library there are several functions where
an unnecessarily complex coding has been used:

  - endTransmission: the availability of data is already
    checked in while(...), therefore need not be checked
    again in the loop.
  - requestFrom: the for-loop has a predefined and fixed
    number of loops. Therefore a check whether the last
    element has been reached is unnecessary and does not
    add any benefit.

Fixes #20