mattairtech/ArduinoCore-samd #22: Windows Max Path length being exceeded.

hex86, 2018-10-06:
I have noticed an error on the Arduino windows platform running the latest beta where the path length can exceed the maximum length allowed by Windows. The problem appears to be with respect to all the option names being part of the cached object file name. The first build will generally work properly but the second time you build the project Arduino will complain that it can't find a file at the path relative to the build folder. The instance I had today the full path was around 275 characters in length while Windows allows only 260 (including the null terminator) I believe. The immediate workaround is restart the IDE each time OR adjusting the option names (I quickly went with the latter).

per1234, 2018-10-08:
Which version of the Arduino IDE are you using?

The problem you report has been resolved by https://github.com/arduino/arduino-builder/commit/ab41198a9a3a9da3e04ca493fd245d0e87b4785c. That fix is in the arduino-builder 1.4.1 used by Arduino IDE 1.8.7.

hex86, 2018-10-17:
I on 1.8.5.