mattairtech/ArduinoCore-samd #28: Wire library on MT-D11/SAMD11 compiles too big for SAMD11 memory

LowPowerLab, 2019-04-19:
Perhaps related and a follow up to Issue #17 . This is for the MT-D11 board / SAMD11.
Main issue: **Using Wire on MT-D11/SAMD11 is basically impossible.**

So I wonder if you have any tips on reducing compile size of I2C/Wire code to something that is usable, and leaves space for actual application code.

**Details of what I'm doing:**

I'm trying to compile this [simple example](https://github.com/tockn/MPU6050_tockn/blob/master/examples/GetAngle/GetAngle.ino) that reads from an I2C sensor. I followed all the optimizations in #17 to reduce the compiled size to a minimum. Still the size overflows by 3K (19K in total) and that's without bootloader, no USB and hence no way to do simple serial debugging.

**The same example compiled for a UNO/328p is less than 10K!**

I tried it on a SAMD21 and there the 10-15K compile size of the Wire is negligible given the huge 256k memory. So it's never noticeable. On the SAMD11 however, as soon as 1 call to anything that calls Wire is made, the compiled sketch size blows up.

FWIW - going from bare minimum sketch to adding Serial, delay increases size from around 1K to over 4K (can't remember exact numbers).

I realize this may be because of bulky dependencies inherited from arduino/atmel, which this core probably relies upon. I haven't spent too much time tracking down what actually could be done.

mattairtech, 2019-05-28:
I compiled the examples/digital_potentiometer.ino demo that comes with Arduino using the D11 with USB and serial disabled and it compiled to 4KB. Looking at the verbose output (go to File->Preferences and tick the checkbox next to 'Show verbose output during: compilation'), the Wire code uses about 1.2KB (and about 180 bytes + of SRAM). I then loaded the example you referenced and compiled with a D21 so that I could get the verbose output, and it looks like the library the example depends on uses floating point math (note all of the math functions like __aeabi_ddiv and __aeabi_fmul, which come from the standard C math libraries provided by ARM), which takes up many KB of space. Using floating point on a D11 is possible, but leaves very little space. You will need to either find a different library that doesn't use floating point or maybe you can modify the library to avoid the use of floating point.

mattairtech, 2019-12-10:
I will make changes so that an error message is generated if there is floating point code in the sketch or any included libraries.