mattairtech/ArduinoCore-samd #37: Arduino INPUT_PULLUP on pin9 of genericSAMD11C14

roberthart56, 2020-04-09:
When using Arduino IDE, setting INPUT_PULLUP on pin 9 of the generic SAMD11C14A leaves pin 9 floating, but pulls up pin 8.  Other pins seem to work as expected.

your-friend-alice, 2022-08-18:
## Cursed workaround:

re-run `pinMode()` immediately before calling `digitalRead()`, (or, I suspect, just after every call to `Serial.print()` and probably other `Serial` methods I haven't enumerated). From the digging i did, it appears pin 9 is set up as a serial port, so I'm guessing what happens is, every time a `Serial` method is called, it sets the `PMUXEN` bit in the `PINCFG` register for that pin (see [datasheet](https://ww1.microchip.com/downloads/aemDocuments/documents/OTH/ProductDocuments/DataSheets/Atmel-42363-SAM-D11_Datasheet.pdf) section _22.8.13_ for that register description). The only reason it keeps working as an input is that it appears to be mapped to a serial input. Sadly I don't know enough about Arduino to know how to properly fix this, perhaps disabling the right serial port would help.

Slight speed optimization over `pinMode()` workaround if it matters: to reset the `PMUXEN` bit, you can run this instead and save on the other register accesses `pinMode()` does.
```
PORT->Group[0].PINCFG[pin].reg=(uint8_t)(PORT->Group[0].PINCFG[pin].reg & B11111110);
```
with `pin` being the pin's index on Port A, rather than the Arduino pin number (this snippet is hardcoded to Port A since that's the only one the SAMD11C14A has anyways). This reads in the current PINCFG register, sets the `PMUXEN` bit to 0, and writes it back out. If you want to just trust your assumption that `INEN` and `PULLEN` should be set to enabled, and `DRVSTR` should be set to "normal" (this is what `pinMode()` does by default), then this would save on that read and just write out a fixed value:
```
PORT->Group[0].PINCFG[pin].reg=(uint8_t)B00000110;
```

This still isn't great though, because the pullup is still being turned off every time `Serial.print()` is called. Depending on what that pin is hooked up to, periodically glitching the pullup off and back on again (especially if you turn it back on right before reading the input) could cause problems. a more permanent workaround (or an actual fix) is still needed.

ictq-oy, 2022-11-04:
This is because pin 8 and pin 9 are designated as LEDs to be used for Serial or USB in the variant.h of Generic_D11C14A.

If you edit variant.h and comment out the relevant line, you can use them as normal I/O. Specifically, it is around line 131 to 144.