mattairtech/ArduinoCore-samd #43: [Question] I want to ask this before destroy my only uC

klerone, 2021-03-20:
Hello @mattairtech 

I have not experience with Atmel ARM micro Controllers but I have been using Arduino and ATMEGAs, I bought a SAMD11C14 for learning purpose.
I do not have an Atmel-ICE, I have a J-Link OB and it works well with STM32, so want to use it to download the code and do not use the boot-loader to save space. 

My Question is: Do I need to upload first a boot-loader file before uploading Arduino code generated .bin file or just upload the .bin file and the code will run?

klerone, 2021-04-06:
Well, I finally decided to try it and run into problems and hope someone can help me to figure out, let me explain.

I successfully connected the MCU to JLink OB and loaded code 2 times, the first time I did not enable the #define PIN_MAP_COMPACT and uploaded the code to blink an LED (nothing happened), then I recompiled the code and uploaded, this time my LED blink and I was happy, then wanted to PWM that LED but I found that could not connect any more.

Have been more than a week since then, I looked for information and even bought a DAP-Link to try if the problem was by clone J-Link, it is not. 
using openOCD I get an error message and points that the SWD pins maybe disable, my MCU is obviously alive and working because if I power it up the LED blink with the last code i uploaded.

I am not an expert developer, mostly a hobbyist but I am diving in the code to find if there is any line that disabled the SWD pins.  

If someone can help me I want to know how can I manage to upload code again, or erase the MCU to recover it. My pourpose is to learn what happened and if can be undone 

This was my settings in the Arduino IDE:
[image]

Robertobit, 2021-04-07:
Hi, 

can you see your J-Link in Atmel/Microchip Studio listed under Tools?
If yes, have a look if you can access the chip from there.

klerone, 2021-04-09:
Well, I found the solution to my problem.

1.  I  did not read the datasheet very well and I missed the 1K PullUp to SWDIO.
2.  To be able to connect using the J-Link OB (clone) using J-Flash Lite need to sync the click (in the software) with a manual reset, now I can upload code as many times as I want. It works.

Then answer to my first question in case someone else come to the same:
Yes, it is possible to upload .bin file from the Arduino IDE using and external software and is not necessary to upload a bootloader before doing that.

In my case I compile the code in Arduino IDE, then in the Sketch menu click generate the bin file, using J-Flash Lite can upload the code easily even with a clone J-Link.

I have been trying to upload the code using openOCD with a DapLink without success.

closed 2021-04-09 by klerone