mattairtech/ArduinoCore-samd #51: JSON link dead, is there an alternative?

vtjballeng, 2023-12-26:
https://www.mattairtech.com/software/arduino/package_MattairTech_index.json is dead. 

Is there an alternative library link?

fdasbfg, 2024-01-03:
https://web.archive.org/web/20230328113917if_/https://www.mattairtech.com/software/arduino/package_MattairTech_index.json

DeqingSun, 2024-01-29:
Not only the json is dead. I can find an archived version of json. The bigger problem is all packages, toolchains, etc, were also hosted on that dead server. 

It seems there is another project https://github.com/wagiminator/SAMD-Development-Boards/tree/main/ArduinoCore-samd was based on this repo. Might worth trying.

tomaskovacik, 2024-01-29:
https://github.com/ElectronicCats/ArduinoCore-samd